evilgeniuslabs/tree-v2 #2: Eliminate Mapped LEDs

terrag42, 2016-11-06:
The tree video looks great! I would like to use this with my project, but I have a different number of LEDs. I would like to use your websockets app to control my LEDs. I have commented out all the patterns that use mapped LEDs. Now I'm getting an error:
Pride2 and colorwaves2 uses "levelCount" variable.
The compiler gives the error:
'levelCount' was not declared in this scope

What kind of variable is it, and where should I declare it?

terrag42, 2016-11-06:
I was able to get it to compile by declaring two new global variables:
uint16_t levelCount;
uint8_t levels[NUM_LEDS];

jasoncoon, 2016-11-06:
Yeah, if you're not using any of the mapping, you can just remove the Pride2 and ColorWaves2 patterns.

closed 2016-11-06 by jasoncoon