evilgeniuslabs/tree-v2 #5: Analog Read Problem

terrag42, 2016-11-16:
I've been able to get some of my own patterns woven into the tree-v2 code, and I love the fast response that websockets provides.
I'm trying to get some sound reactive patterns using an MSGEQ7. I've got the code (below) working with Juergen Bruegl's webserver code but not with the tree-v2 code.

```
void equalize(){
  digitalWrite(msg7RESET, HIGH);                  // reset the MSGEQ7's counter
  delay(5);
  digitalWrite(msg7RESET, LOW);
  Serial.println("Low"); //for debugging
        for (int x = 0; x < 7; x++){
                digitalWrite(msg7Strobe, LOW);          // output each DC value for each freq band
                delayMicroseconds(35); // to allow the output to settle
                int spectrumRead = analogRead(A0);//returns a value from 0 to 1023
                int MSGval = map(spectrumRead, 0, 511, 0, 255);  // scale analogRead's value to Write's 255 max
               if (MSGval < 50) MSGval = MSGval / 2;                // bit of a noise filter, so the LEDs turn off at low levels
               leds[x]=CHSV(MSGhue,255,MSGval);
               if (millis() - ticker > 100){
                  MSGhue++; //increases hue vale every 100 milliseconds
                ticker = millis();
               }
                digitalWrite(msg7Strobe, HIGH);
                //Serial.println(MSGval);
                Serial.println(spectrumRead);
        }
}
```
Looking at what's coming from the serial monitor, It looks like the ESP's analog port is reading, but the MSGEQ7 is not putting out any voltage.
Here is the serial monitor output:
```
Low
5
3
3
3
3
3
3
Low
4
4
3
3
3
3
3
Low
4
3
3
3
3
3
3
Low
5
3
3
4
3
3
4
```

Any suggestions?

jasoncoon, 2016-11-16:
Cool!  I welcome pull requests if you'd like to share any patterns, improvements, etc.

So, the MSGEQ7 works fine with Juergen's sketch, but not tree-v2?  All with the exact same hardware?

I have a few MSGEQ7 boards laying around, and I've been meaning to get one hooked up and working with the tree.  I'll see what I can do.  The last time I tried it (with an Adafruit Huzzah ESP8266 board), I was unable to get it working because the the board's analog input is only 1V.  I think the Wemos D1 Mini's analog input is 3.2V, so I'll try again with it.

terrag42, 2016-11-18:
I am able to use Juergen's sketch with the exact same hardware and get the A0 pin (Wemos D1 mini) to read voltages correctly. You're right the Wemos D1 mini has a voltage divider on the ADC pin. It's a 220k on the input and a 100k pull-down. This effectively changes a 0-3.3V analog input into the 0-1V that is readable by the ESP. You don't need to wire up an MSGEQ7 to test the analogRead() function. Just a potentiometer connected to 3.3V on one side, ground on the other side and the wiper connected to A0. Make a simple function that prints the analog read value to the serial monitor. it should be between 0 and 1023.