evilgeniuslabs/tree-v2 #8: Dynamic Fields not working

terrag42, 2016-11-18:
Hello, I am having trouble getting the dynamic fields to display. The only field working right now is "speed". 
On Chrome on my android phone, the app is giving an error where the status message usually displays. The error "Fail: [object Object] parsererror" appears after I select a solid color. 
Should the dynamic field names be changing to orange in the Arduino IDE when they are typed?
I uploaded the sketch, then used the ESP8266 sketch data upload function to upload the data.
Adruino IDE 1.6.12
esp8266 core version 2.3.0
Wemos D1 Mini

jasoncoon, 2016-11-18:
Made some changes today, working fine here.  If you're working from your fork, try pulling the latest from master.

terrag42, 2016-11-18:
I opened the issue shortly after pulling the latest from master.

jasoncoon, 2016-11-18:
Just want to verify: are you using your fork or mine?  Your fork is behind 6 commits behind mine: https://github.com/terrag42/tree-v2/pull/1

terrag42, 2016-11-18:
I'm using your fork. I am getting the "Fail: [object Object] parsererror" only when using a modified tree-v2.ino sketch where I've removed all the mapped patterns. I also uploaded an unmodified tree-v2.ino sketch just to try the web app, but I'm still not seeing any dynamic fields in either sketch.

jasoncoon, 2016-11-18:
Can you use Chrome Dev Tools and look at the responses coming back from the ESP8266 web server?

You shouldn't see any errors in the console.

Here's what I see:

[image]

[image]

The request for /all should return the following data:

```
[
  {
    "name": "power",
    "label": "Power",
    "type": "Boolean",
    "value": 0
  },
  {
    "name": "brightness",
    "label": "Brightness",
    "type": "Number",
    "value": 255
  },
  {
    "name": "pattern",
    "label": "Pattern",
    "type": "Select",
    "value": 15,
    "options": [
      "Pride",
      "Pride 2",
      "Color Waves",
      "Color Waves 2",
      "Northward Rainbow",
      "Northeastward Rainbow",
      "Eastward Rainbow",
      "Southeastward Rainbow",
      "Southward Rainbow",
      "Southwestward Rainbow",
      "Westward Rainbow",
      "Northwestward Rainbow",
      "Rotating Rainbow",
      "Outward Rainbow",
      "Inward Rainbow",
      "Falling Rainbow",
      "Rising Rainbow",
      "Rotating Palette",
      "Outward Palette",
      "Inward Palette",
      "Falling Palette",
      "Rising Palette",
      "Rainbow Twinkles",
      "Snow Twinkles",
      "Cloud Twinkles",
      "Incandescent Twinkles",
      "Rainbow",
      "Rainbow With Glitter",
      "Solid Rainbow",
      "Confetti",
      "Sinelon",
      "Beat",
      "Juggle",
      "Fire",
      "Water",
      "Solid Color"
    ]
  },
  {
    "name": "speed",
    "label": "Speed",
    "type": "Number",
    "value": 55
  },
  {
    "name": "autoplay",
    "label": "Autoplay",
    "type": "Section"
  },
  {
    "name": "autoplay",
    "label": "Autoplay",
    "type": "Boolean",
    "value": 0
  },
  {
    "name": "autoplayDuration",
    "label": "Autoplay Duration",
    "type": "Number",
    "value": 1
  },
  {
    "name": "solidColor",
    "label": "Solid Color",
    "type": "Section"
  },
  {
    "name": "solidColor",
    "label": "Color",
    "type": "Color",
    "value": "255,0,0"
  }
]
```

terrag42, 2016-11-19:
[image]
 When I request /all, I get something the following
`[{"name":"power","label":"Power","type":"Boolean","value":1},{"name":"brightness","label":"Brightness","type":"Number","value":126},{"name":"pattern","label":"Pattern","type":"Select","value":5,"options":["Pride","Color Waves","Rainbow Twinkles","Snow Twinkles","Cloud Twinkles","Incandescent Twinkles","Rainbow","Rainbow With Glitter","Solid Rainbow","Confetti","Sinelon","Beat","Juggle","Fire","Water","Solid Color"]},{"name":"speed","label":"Speed","type":"Number","value":16},{"name":"autoplay","label":"Autoplay","type":"Section"},{"name":"autoplay","label":"Autoplay","type":"Boolean","value":1},{"name":"autoplayDuration","label":"Autoplay Duration","type":"Number","value":22},{"name":"solidColor","label":"Solid Color","type":"Section"},{"name":"solidColor","label":"Color","type":"Color","value":"79,129,255"}]`

jasoncoon, 2016-11-19:
OK, figured it out, pull the latest from master.  I still had the field count hard-coded to 9, then added the field separators...  :)  Switched to automatically setting the field count and now they show up.

I was getting the same error when setting a solid color, until I updated the sketch earlier to return text/plain instead of text/json: https://github.com/terrag42/tree-v2/pull/1/commits/766d00e385533ca57290679870866bc9b7376875#diff-c83915d07a987682fa00b85bbb126601L419

jasoncoon, 2016-11-19:
Oh, also, are you using your device's IP address or mDNS?  I'm seeing significantly faster response when using the IP.  mDNS lookups are taking several seconds on my network.