evilgeniuslabs/tree-v2 #11: Add Scheduler

mbelles, 2016-11-22:
Investigate ways of scheduling off and on. While this may be complete overkill, it would be fun to have it kick on and off for a few hours in the morning, and again at night. Even something as simple as an external time request and mapping would do it. (i.e., on for an hour in the morning, and a few hours in the evening)

mbelles, 2016-11-22:
Windows/Quartz scheduler => REST API to power on/off.

dmacias72, 2016-12-02:
If you remove "HTTP_POST" from /power by changing this line

` webServer.on("/power", HTTP_POST, []() {`

to

` webServer.on("/power", []() {`

you can turn them on or off with

http://ipaddress/power?value=1 for on and
http://ipaddress/power?value=0 or http://ipaddress/power for off

You can then control it through script or home automation software.  You can also do the same for brightness, pattern, speed, autoplay and autoplay duration. Maybe I'll make a pull request.

mbelles-firemon, 2016-12-03:
GET requests should never change the stage of the server. It is a POST by design.

There are plenty of ways to script a request with a different HTTP method, such as wget or curl. There is already an example in code: https://github.com/evilgeniuslabs/tree-v2/blob/master/power.sh

I've left this ticket open, in case I get time and decide to do something more advanced.

dmacias72, 2016-12-03:
I GET it :) I figured out how to POST with lua. In case anyone wonders how
`local http = require("socket.http")

http.TIMEOUT = 5

result, status = http.request("http://192.168.1.34/power?value=0", "run=run")`

viktor6, 2020-04-10:
Yes, how does it work from a web browser?

jasoncoon, 2020-04-10:
@viktor6 use the web ui.  For controlling it without the web ui, see the answers to your issue here: https://github.com/evilgeniuslabs/tree-v2/issues/29

I will not be implementing a scheduler UI any time soon.  I welcome pull requests from anyone that wants to do it.  I'm closing this issue.

closed 2020-04-10 by jasoncoon