evilgeniuslabs/tree-v2 #12: Cannot compile under current Arduino IDE (map.h)

kelexel, 2016-11-28:
Hello!

I am using the current Arduino IDE, I installed all the libs specified in the README by downloading them and placing them to my arduino libraries directory.

I downloaded the current tree-v2, renamed `tree-v2-master` to `tree-v2`
I edited *_PIN in the sketch to reflect my board's layout (using an ESP-01 here).

But, when running the sketch before compiling:

```
n file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:21: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
In file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:25: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
In file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:29: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
Map.h:89: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
In file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:108: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
In file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:112: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
In file included from /Users/rud/Downloads/tree-v2/tree-v2.ino:71:0:
Map.h:116: error: too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'
 };
 ^
exit status 1
too many initializers for 'const uint8_t [64] {aka const unsigned char [64]}'

```

mbelles, 2016-11-28:
Mind sharing a link to your code changes in a commit? (i.e., fork => branch => commit => share link) Seeing the code changes would make it much easier to suggest fixes, otherwise you're likely somewhat on your own with broken code changes.

jasoncoon, 2016-11-28:
Looks like you might have changed NUM_LEDS from 250 to 64.  All of the arrays in Map.h hold coordinates for 250 LEDs.  You'll need to redo the maps, or comment them out and then comment out all of the patterns that use the maps.  I highly recommend just starting with [esp8266-fastled-webserver](https://github.com/jasoncoon/esp8266-fastled-webserver), which doesn't have any special LED maps.

kelexel, 2016-11-29:
@jasoncoon aha, yes, your are right, I only use 64 LEDs on this setup..
I tried esp8266-fastled-webserver, and ended up porting it to ESPAsyncWebServer remember ? (https://github.com/kelexel/ESPDeskLamp) ;)
Was just curious to test your latest changes...
Now I understand the issue, sorry for the troubles (maybe add a note about not modifying the number of LEDs in the README ? ;)

@mbelles: I didn't modify a thing, aside, the number of LEDs, hence the bug.

closed 2018-01-14 by jasoncoon