evilgeniuslabs/tree-v2 #17: Not found: /

TerroBladeZ, 2017-12-05:
After following the steps I am getting Not found: / on the ip address which comes up on the serial monitor.
Getting errors as below while uploading

In file included from C:\Users\cryst\Downloads\tree-v2-master\tree-v2\tree-v2.ino:22:0:

C:\Users\cryst\Documents\Arduino\libraries\FastLED-master/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.006

 #    pragma message "FastLED version 3.001.006"

                     ^

In file included from C:\Users\cryst\Documents\Arduino\libraries\FastLED-master/FastLED.h:68:0,

                 from C:\Users\cryst\Downloads\tree-v2-master\tree-v2\tree-v2.ino:22:

C:\Users\cryst\Documents\Arduino\libraries\FastLED-master/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

In file included from C:\Users\cryst\Downloads\tree-v2-master\tree-v2\tree-v2.ino:41:0:

sketch\Field.h:37:1: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

 ^

mariusmotea, 2017-12-05:
Not sure if evilgenius is still maintaining this great software, but these pragma "errors" are not errors. This is my successful build output:

```
In file included from /home/marius/Documents/tree-v2/tree-v2.ino:21:0:
/home/marius/Arduino/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.006
 #    pragma message "FastLED version 3.001.006"
                     ^
In file included from /home/marius/Arduino/libraries/FastLED/FastLED.h:68:0,
                 from /home/marius/Documents/tree-v2/tree-v2.ino:21:
/home/marius/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Sketch uses 318717 bytes (30%) of program storage space. Maximum is 1044464 bytes.
Global variables use 48396 bytes (59%) of dynamic memory, leaving 33524 bytes for local variables. Maximum is 81920 bytes.
```
One issue i had was with D7 pin specified by default, this is working only if you select the board Wemos D1

TerroBladeZ, 2017-12-05:
Now i am getting similar output. But not being able to see the webpage. Whrn i type my IP it gives me bunch of numbers. I Dont know what i am doing wrong.

mariusmotea, 2017-12-05:
you did not upload the html files.

add this plugin to your arduino: https://github.com/esp8266/arduino-esp8266fs-plugin

> Open a sketch (or create a new one and save it).
> Go to sketch directory (choose Sketch > Show Sketch Folder).
> Create a directory named data and any files you want in the file system there.
> Make sure you have selected a board, port, and closed Serial Monitor.
> Select Tools > ESP8266 Sketch Data Upload menu item. This should start uploading the files into ESP8266 flash file system. When done, IDE status bar will display SPIFFS Image Uploaded message. Might take a few minutes for large file system sizes.

mariusmotea, 2017-12-05:
Also, something very important, before you flash the firmware and the files you must wipe entire spi flash.
ex: `esptool.py --port /dev/ttyUSB0 erase_flash`

TerroBladeZ, 2017-12-05:
Finally webpage worked and I used Datapin 15 Now the problem is my lights are blinking random color and not responding to the presets or the solid color. I am using Adafruit Huzzah and WS8211 Pixel LEDs

Blink515, 2017-12-05:
How do you have things wired up?   Are you powering with an external power
supply or directly from the Huzzah?  How big of a power supply are you
using?  Are your grounds connected between LEDs, Huzzah and power supply?
I would also take a wild guess and say that you need to use a level
shifter.  The Huzzah is 3.3V logic and WS2811 LEDs expect around 5V.  Yes,
sometimes it works without a level shifter but not always.

TerroBladeZ, 2017-12-06:
I am  powering LED using  an external power supply which is 5V20A. 
grounds are not connected between LEDs and Huzzah, but are connected with led to powersupply. My Huzzah is getting power from PC
Where to add a level shifter? 
Also i have an resistor of 330OHM connected to dataline between huzzah and led

mariusmotea, 2017-12-06:
How many leds do you have? If more than 250, try one time by setting the number of leds to 250. I have 300 and lot of paterns are not working. However few of them that are working are wonderful, especially Confeti. If i setup the number of leds to 250 all paterns are working.

TerroBladeZ, 2017-12-06:
Finally everything works. I have 100 for testing will be adding 400 more. This time I plugged + & - wires to the Esp instead of power supply. Do you have any suggestion on wire connection?

mariusmotea, 2017-12-06:
i'm working on a python script that will generate map config for strips based on number of circles and number of leds/circle. Is a little complicated, but in the end i will finish.

TerroBladeZ, 2017-12-07:
Can you please share the code when your done?

mariusmotea, 2017-12-07:
I fork this repo and upload generate_map_h.py file. Remember this is made for strips. I push some quick updates also in tree-v2.ino to support more than 255 leds. More changes must be made to have all patterns working, but with what i made most of them will work. If somebody else found other fixes please let me know.

closed 2018-01-14 by jasoncoon