evilgeniuslabs/tree-v2 #18: Most of patterns are just solid colors if using 300 leds.

mariusmotea, 2017-12-05:
Hi,

I have a 5m strip with 300 leds. i setup NUM_LEDS to this values and i saw that most of the patterns are just solid color. With this problem the autoplay future become useless, most of the time showing just  one color. If i compile the code with NUM_LEDS set to 250 they are working fine, but the head of my tree remain in dark. Also 5 meters is not that much for a 1.5 meter tree, so i cannot cut the extra 50 leds because i will not be able to cover entire tree anymore.

Blink515, 2017-12-06:
Any patterns/animations using uint8_t to set the LED data will not work with more than 255 LEDs. You'll need to update the code to use uint16_t or just int.  Make sure you don't change it everywhere because it is needed for palette index and hue since they max out at 255 and will roll back over to 0.  Hopefully that makes sense.

mariusmotea, 2017-12-06:
I finish my script that generate Map.h file for strip automatically. Here i believe the values are correct. problem is that  tree-v2 was not designed to work with more than 255 leds, i replaced uint8_t with  uint16_t in lot of places and still some effects are not working. Currently i will say that 80% are working. I will share my script, maybe others will use it.

funel, 2017-12-07:
I have the same issue. I have 300 leds and can't make it work. when using 250 leds the program works very well.
Can anybody tell where to replace unit8_t with unit16_t exactly?

mariusmotea, 2017-12-07:
Check my fork. This is not 100% done, but lot of patters with start to work. In next days i will improve the code.

Blink515, 2017-12-07:
You'll need to replace uint8_t with uint16_t wherever you see
`for (uint8_t i = 0; i < NUM_LEDS; i++)` or `for (uint8_t i =NUM_LEDS-1; i >0; i--)` 

After you get them replaced, go through each animation and test them.  If you run into any other issues just report which animation(s) is still giving problems.

funel, 2017-12-07:
thanks for your prompt reply.
do I have to look into tree-v2.ino file or map file?
this is my first arduino and first led project, so sorry if I'm asking stupid questions ;)

Blink515, 2017-12-07:
Look in the .ino file.

funel, 2017-12-10:
Thanks a lot for your support and a special thank to **mariusmotea** for his updated code.
here is my tree with 300 leds: https://www.youtube.com/edit?o=U&video_id=VJmHQcgkv9E

jasoncoon, 2018-01-14:
That link doesn't work for me, but this one does: https://www.youtube.com/watch?v=VJmHQcgkv9E&t=23s

Looks great, glad you got it working!

closed 2018-01-14 by jasoncoon