evilgeniuslabs/tree-v2 #21: wdt reset when use it TwinkleFOX patterns

Bibbbi, 2017-12-23:
The problem is that the module will reset due to wdt after used TwinkleFOX patterns.

e.g. Retro C9 Twinkles, Red & White Twinkles, Blue & White Twinkles, Red, Green & White Twinkles, Fairy Light Twinkles, Snow 2 Twinkles, Holly Twinkles, Ice Twinkles, Party Twinkles, Forest Twinkles, Lava Twinkles, Fire Twinkles, Cloud 2 Twinkles and Ocean Twinkles
```
wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```
Workarround Fix (**TwinkleFOX.h** in Line 197):

search for: `for(uint8_t i = 0; i < NUM_LEDS; i++) {`
change to: `for(uint16_t i = 0; i < NUM_LEDS; i++) {`

eziologico, 2017-12-28:
It is not a workaround, it is the real solution to fix it if you have more than 255 LED. I have 600 LED (2 Strip of 300 LED).

Even "Palette Waves" is affected by this problem.
At the end of the .ino file change to "uint16_t i = ..." in the second "for" of the paletteWaves() function like below:

`void paletteWaves() {
  fadeToBlackBy( leds, NUM_LEDS, 20);

  for (uint8_t levelIndex = 0; levelIndex < levelCount; levelIndex++) {
    for (uint16_t i = levelStart[levelIndex]; i < levelEnd[levelIndex]; i++) {
      leds[i] = ColorFromPalette(gCurrentPalette, gHue + levelIndex, beatsin8(speed + levelIndex));
    }
  }`

So it is a perfect project now!

eziologico, 2017-12-28:
there is still a reboot problem, after few time, due to the following error:

Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4020db65, epc2=0x00000000, epc3=0x00000000, excvaddr=0xbdc44716, depc=0x00000000

Exception (9):
epc1=0x4020db65 epc2=0x00000000 epc3=0x00000000 excvaddr=0xbdc44716 depc=0x00000000

ctx: cont 
sp: 3fff4030 end: 3fff4290 offset: 01a0

>>>stack>>>
3fff41d0:  3f00006c 0000003f 3fff4218 4010068c  
3fff41e0:  00000258 3fff1286 00000257 3fff218d  
3fff41f0:  3fff102e 3fff102d 3fff420c 3fff2ecc  
3fff4200:  000000eb 3fff2ed8 3fff2ecc 4020d230  
3fff4210:  3fffdad0 3fff11a0 3fff19f0 4020b132  
3fff4220:  33000000 00660000 00009900 ff0000cc  
3fff4230:  33ff0000 0066ff00 ff0099ff ffff00cc  
3fff4240:  33ffff00 ff66ffff ffff99ff 3fff3264  
3fff4250:  3fffdad0 3fff11a0 3fff19f0 4020aed8  
3fff4260:  00000000 00000000 00000001 4021537d  
3fff4270:  3fffdad0 00000000 3fff325d 402153a8  
3fff4280:  feefeffe feefeffe 3fff3270 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮⸮
Heap: 23072
Boot Vers: 31
CPU: 160
SDK: 1.5.3(aec24ac9)
Chip ID: 2047092
Flash ID: 1589487
Flash Size: 16777216
Vcc: 65535

FS File: /favicon.ico, size: 4286
FS File: /css/bootstrap.min.css, size: 121093
FS File: /css/styles.css, size: 31
FS File: /css/simple.css, size: 237
FS File: /css/jquery.minicolors.min.css, size: 96597
FS File: /images/atom196.png, size: 5469
FS File: /js/r-websocket.min.js, size: 3100
FS File: /js/simple.js, size: 6261
FS File: /js/bootstrap.min.js, size: 37045
FS File: /js/jquery.minicolors.min.js, size: 15463
FS File: /js/jquery-3.1.1.min.js, size: 86709
FS File: /js/app.js, size: 12933
FS File: /index.htm, size: 11096
FS File: /edit.htm, size: 7887
FS File: /fonts/glyphicons.svg, size: 108738
FS File: /fonts/glyphicons.ttf, size: 45404
FS File: /fonts/glyphicons.eot, size: 20127
FS File: /fonts/glyphicons.woff2, size: 18028
FS File: /fonts/glyphicons.woff, size: 23424
FS File: /simple.htm, size: 2203

jasoncoon, 2017-12-31:
Sorry for the trouble, and for the delayed response.  I've been out for the holidays.  :)

I just integrated these and another fix: https://github.com/evilgeniuslabs/tree-v2/commit/48b11c432b8f2f9807ce13e25b89c8ab2f93257c

Let me know if you find any others.  Thanks!

closed 2018-01-14 by jasoncoon