evilgeniuslabs/tree-v2 #22: Reset Loop with: ets Jan  8 2013,rst cause:2, boot mode:(3,6)

atatistcheff, 2017-12-30:
Just got this installed and running and was going through a few patterns when it started into a "Soft WDT reset" loop.  I'm using a Wemos D1 mini with just 100 WS2811s to test with.  I've tried re-flashing the sketch and data and am still seeing the same issue.  

I tried uploading the "blink" demo script to the D1 and it works fine, no soft resets.

The serial output is below.  This repeats continually about every 3 seconds:

Heap: 26816
Boot Vers: 31
CPU: 160
SDK: 1.5.3(aec24ac9)
Chip ID: 6742563
Flash ID: 1458376
Flash Size: 4194304
Vcc: 65535

FS File: /css/bootstrap.min.css, size: 121093
FS File: /css/jquery.minicolors.min.css, size: 96597
FS File: /css/simple.css, size: 237
FS File: /css/styles.css, size: 31
FS File: /edit.htm, size: 7887
FS File: /favicon.ico, size: 4286
FS File: /fonts/glyphicons.eot, size: 20127
FS File: /fonts/glyphicons.svg, size: 108738
FS File: /fonts/glyphicons.ttf, size: 45404
FS File: /fonts/glyphicons.woff, size: 23424
FS File: /fonts/glyphicons.woff2, size: 18028
FS File: /images/atom196.png, size: 5469
FS File: /index.htm, size: 11096
FS File: /js/app.js, size: 12933
FS File: /js/bootstrap.min.js, size: 37045
FS File: /js/jquery-3.1.1.min.js, size: 86709
FS File: /js/jquery.minicolors.min.js, size: 15463
FS File: /js/r-websocket.min.js, size: 3100
FS File: /js/simple.js, size: 6261
FS File: /simple.htm, size: 2203

Hostname: Tree-ESP8266-66e223
Connecting to Wonkavision
HTTP web server started
Web socket server started

Soft WDT reset

ctx: cont 
sp: 3fff3310 end: 3fff3550 offset: 01b0

>>>stack>>>
3fff34c0:  0000005b 000071a9 00000568 40209139  
3fff34d0:  00e5c200 00000000 3fff1a00 40211e80  
3fff34e0:  00000239 00000000 3fff1a20 4020f2b9  
3fff34f0:  3ffe9dd0 00000000 000003e8 3fff2528  
3fff3500:  0000000a 3fff09d0 3fff0974 402092bd  
3fff3510:  3fffdad0 3fff09d0 3fff0974 40209fd9  
3fff3520:  feefeffe feefeffe feefeffe 3fff2528  
3fff3530:  3fffdad0 00000000 3fff2521 40213ad8  
3fff3540:  feefeffe feefeffe 3fff2530 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112

atatistcheff, 2017-12-30:
BTW, this behaves the same on USB power from my Macbook with nothing connected to D7 as well as an external 5v (measured to verify it's really 5v) 2000ma power supply.  Also tried a 4700uf decoupling capacitor on the 5v to verify it's not a power issue.

mariusmotea, 2017-12-30:
You are using more than 255 leds? If yes, check other issues. If you have problems with only a few patterns, just comment them in "PatternAndNameList patterns " array.

atatistcheff, 2017-12-30:
I am using the stock code from Github.  The only change I've made is to the WiFi SSID and password.  The code already has NUM_LEDS set to 300.  This issue occurs with 100 pixels as well as zero pixels.  Basically as soon as the code is finished uploading to the D1 the loop begins.

mariusmotea, 2017-12-30:
Before you upload the firmware you must wipe the spi flash
`esptool.py --port /dev/ttyUSB0 erase_flash`
on my fork of this repo is my code that works with 300 leds. Is not perfect because i fix some issues by replacing lot of uint8_t with uint16_t, some of them i don't believe must be replaced and are just some extra ram consuming. Also i comment fire and water patterns that are still causing device reset and i was not able to fix them, but all other patterns and functions are working.

atatistcheff, 2017-12-30:
Thanks for the command to clear the SPI flash that will come in handy!  

I'm kind of still in the process of figuring out Github but it looks to me like there's only one branch for this project and the 300 LED fix is the most recent version.  Is there actually a fork for the 300 LED support or was the main project modified?  I'm just asking because if that's the case it would be nice to have some kind of note that the project will not work fully without some code modifications.  I mean at a minimum the Readme should state that your ESP8266 will start resetting after a few patterns.  Otherwise it's not really obvious why all of a sudden the device starts resetting.

Overall it's a really impressive project though!  Incredible what you've gotten out of a tiny ESP8266 module!

mariusmotea, 2017-12-30:
I believe the code is working fine with less than 250 leds, but the installation instructions can be improved. I had similar issues in the past and i was know that some projects require spi flash to be erased because of the file storage. 
my fork is [this](https://github.com/mariusmotea/tree-v2), as i said, is not perfect, but is working for me.

atatistcheff, 2017-12-30:
Ok just to be clear I'm not talking about your fork in this thread.  The code at https://github.com/evilgeniuslabs/tree-v2 (which is not your fork) is set to 300 LEDs and fails for some patterns.  That's the crux of this issue.

eziologico, 2017-12-31:
Did you fixed the 2 missing uint16_t as explained in issue #21?
My code plays all the pattern now with 600 LED, even if the reboot problem is stil present.

mariusmotea, 2017-12-31:
@eziologico can you provide your code? The best will be to fork this repo and create a PR.

eziologico, 2017-12-31:
Of course, this code is for two 300 LED strip in line.
I espanded the Map.h file for 600 LED, one strip on level0 and the other on level1.

[Strip_600.zip](https://github.com/evilgeniuslabs/tree-v2/files/1595420/Strip_600.zip)

eziologico, 2017-12-31:
Yes, you are right, I just discovered that Fire and Water does not work for me too and the board reboot.
We must investigate!

mariusmotea, 2017-12-31:
i comment them in order to play all patterns in loop, otherwise it will remain stuck there and play only last working pattern

jasoncoon, 2017-12-31:
Sorry for the trouble, and for the delayed response.  I've been out for the holidays.  :)

I just fixed a few issues: https://github.com/evilgeniuslabs/tree-v2/commit/48b11c432b8f2f9807ce13e25b89c8ab2f93257c

Let me know if you find any others.  Thanks!

mariusmotea, 2017-12-31:
`heat[NUM_LEDS];` did the trick for Fire and Water patterns. Only one small issue remain, meteors are placed only in the first 255 leds, in my case with 300 leds is not such a big problem.

eziologico, 2018-01-03:
In the Meteors.h file are other "uint16_t" missing, but it seams that the level 0 is not involved. LED on level 0 never goes on, only LED 0 every time as first LED.

I have 250 LED and 11 levels on my XTree, from 0 to 10.

```
Heap: 26128
Boot Vers: 31
CPU: 160
SDK: 1.5.3(aec24ac9)
Chip ID: 2047092
Flash ID: 1589487
Flash Size: 16777216
Vcc: 65535

FS File: /favicon.ico, size: 4286
FS File: /css/bootstrap.min.css, size: 121093
FS File: /css/styles.css, size: 31
FS File: /css/simple.css, size: 237
FS File: /css/jquery.minicolors.min.css, size: 96597
FS File: /images/atom196.png, size: 5469
FS File: /js/r-websocket.min.js, size: 3100
FS File: /js/simple.js, size: 6261
FS File: /js/bootstrap.min.js, size: 37045
FS File: /js/jquery.minicolors.min.js, size: 15463
FS File: /js/jquery-3.1.1.min.js, size: 86709
FS File: /js/app.js, size: 12933
FS File: /index.htm, size: 11096
FS File: /edit.htm, size: 7887
FS File: /fonts/glyphicons.svg, size: 108738
FS File: /fonts/glyphicons.ttf, size: 45404
FS File: /fonts/glyphicons.eot, size: 20127
FS File: /fonts/glyphicons.woff2, size: 18028
FS File: /fonts/glyphicons.woff, size: 23424
FS File: /simple.htm, size: 2203

Hostname: Tree-ESP8266-1f3c74
Connect to Wi-Fi access point: ESP8266-3C74
and open http://192.168.4.1 in your browser
HTTP web server started
Web socket server started
add 1
aid 1
station: a8:5b:78:08:5a:e9 join, AID = 1
[0] Connected from 192.168.4.2 url: /
Level: 5 Angle: 78 LED: 187
Level: 5 Angle: 78 LED: 187
Level: 4 Angle: 78 LED: 187
Level: 4 Angle: 78 LED: 164
Level: 3 Angle: 78 LED: 164
Level: 3 Angle: 78 LED: 136
Level: 2 Angle: 78 LED: 136
Level: 2 Angle: 78 LED: 101
Level: 1 Angle: 78 LED: 101
Level: 1 Angle: 78 LED: 62
Level: 10 Angle: 139 LED: 0
Level: 10 Angle: 139 LED: 249
Level: 9 Angle: 139 LED: 249
Level: 9 Angle: 139 LED: 246
Level: 8 Angle: 139 LED: 246
Level: 8 Angle: 139 LED: 238
Level: 7 Angle: 139 LED: 238
Level: 7 Angle: 139 LED: 227
Level: 6 Angle: 139 LED: 227
Level: 6 Angle: 139 LED: 211
Level: 5 Angle: 139 LED: 211
Level: 5 Angle: 139 LED: 192
Level: 4 Angle: 139 LED: 192
Level: 4 Angle: 139 LED: 170
Level: 3 Angle: 139 LED: 170
Level: 3 Angle: 139 LED: 143
Level: 2 Angle: 139 LED: 143
Level: 2 Angle: 139 LED: 110
Level: 1 Angle: 139 LED: 110
Level: 1 Angle: 139 LED: 72
Level: 10 Angle: 186 LED: 0
Level: 10 Angle: 186 LED: 249
Level: 9 Angle: 186 LED: 249
Level: 9 Angle: 186 LED: 247
Level: 8 Angle: 186 LED: 247
Level: 8 Angle: 186 LED: 240
Level: 7 Angle: 186 LED: 240
Level: 7 Angle: 186 LED: 229
Level: 6 Angle: 186 LED: 229
Level: 6 Angle: 186 LED: 214
Level: 5 Angle: 186 LED: 214
Level: 5 Angle: 186 LED: 196
Level: 4 Angle: 186 LED: 196
Level: 4 Angle: 186 LED: 174
Level: 3 Angle: 186 LED: 174
Level: 3 Angle: 186 LED: 148
Level: 2 Angle: 186 LED: 148
Level: 2 Angle: 186 LED: 117
Level: 1 Angle: 186 LED: 117
Level: 1 Angle: 186 LED: 79
Level: 10 Angle: 137 LED: 0
Level: 10 Angle: 137 LED: 249
Level: 9 Angle: 137 LED: 249
Level: 9 Angle: 137 LED: 246
Level: 8 Angle: 137 LED: 246
Level: 8 Angle: 137 LED: 238
Level: 7 Angle: 137 LED: 238
Level: 7 Angle: 137 LED: 226
Level: 6 Angle: 137 LED: 227
Level: 6 Angle: 137 LED: 211
Level: 5 Angle: 137 LED: 211
Level: 5 Angle: 137 LED: 192
Level: 4 Angle: 137 LED: 192
Level: 4 Angle: 137 LED: 169
Level: 3 Angle: 137 LED: 169
Level: 3 Angle: 137 LED: 143
Level: 2 Angle: 137 LED: 143
Level: 2 Angle: 137 LED: 110
Level: 1 Angle: 137 LED: 110
Level: 1 Angle: 137 LED: 71
Level: 10 Angle: 38 LED: 0
Level: 10 Angle: 38 LED: 248
Level: 9 Angle: 38 LED: 248
Level: 9 Angle: 38 LED: 244
Level: 8 Angle: 38 LED: 244
Level: 8 Angle: 38 LED: 234
Level: 7 Angle: 38 LED: 234
Level: 7 Angle: 38 LED: 221
Level: 6 Angle: 38 LED: 221
Level: 6 Angle: 38 LED: 205
Level: 5 Angle: 38 LED: 205
Level: 5 Angle: 38 LED: 184
Level: 4 Angle: 38 LED: 184
Level: 4 Angle: 38 LED: 160
Level: 3 Angle: 38 LED: 160
Level: 3 Angle: 38 LED: 131
Level: 2 Angle: 38 LED: 131
Level: 2 Angle: 38 LED: 96
Level: 1 Angle: 38 LED: 96
Level: 1 Angle: 38 LED: 56
Level: 10 Angle: 5 LED: 0
Level: 10 Angle: 5 LED: 248
Level: 9 Angle: 5 LED: 248
Level: 9 Angle: 5 LED: 243
Level: 8 Angle: 5 LED: 243
Level: 8 Angle: 5 LED: 233
Level: 7 Angle: 5 LED: 233
Level: 7 Angle: 5 LED: 219
Level: 6 Angle: 5 LED: 219
Level: 6 Angle: 5 LED: 202
Level: 5 Angle: 5 LED: 202
Level: 5 Angle: 5 LED: 181
Level: 4 Angle: 5 LED: 181
Level: 4 Angle: 5 LED: 156
Level: 3 Angle: 5 LED: 156
Level: 3 Angle: 5 LED: 128
Level: 2 Angle: 5 LED: 128
Level: 2 Angle: 5 LED: 91
Level: 1 Angle: 5 LED: 91
Level: 1 Angle: 5 LED: 51
Level: 10 Angle: 83 LED: 0
Level: 10 Angle: 83 LED: 249
Level: 9 Angle: 83 LED: 249
Level: 9 Angle: 83 LED: 245
Level: 8 Angle: 83 LED: 245
Level: 8 Angle: 83 LED: 236
Level: 7 Angle: 83 LED: 236
Level: 7 Angle: 83 LED: 224
Level: 6 Angle: 83 LED: 224
Level: 6 Angle: 83 LED: 208
Level: 5 Angle: 83 LED: 208
Level: 5 Angle: 83 LED: 188
Level: 4 Angle: 83 LED: 188
Level: 4 Angle: 83 LED: 164
Level: 3 Angle: 83 LED: 164
Level: 3 Angle: 83 LED: 136
Level: 2 Angle: 83 LED: 136
Level: 2 Angle: 83 LED: 102
Level: 1 Angle: 83 LED: 102
Level: 1 Angle: 83 LED: 63
```