evilgeniuslabs/tree-v2 #23: Hello. How can I calculate the X and Y coordinates? What formulas or programs are used? What to read, what would be best to learn how to build a map? I'm studying, and without your help it will be difficult.

and4480, 2018-01-13:


mariusmotea, 2018-01-13:
There is no unique formula, depending on leds placement you need to elaborate your own formula. For examples i create the python script that calculate xyz coordinates, but here was simple pattern (sin, cos) and this is applicable only if leds are placed in circles.
To resume... is just geometry.

and4480, 2018-01-13:
I do not understand how the number x256 y256 in this table https://docs.google.com/spreadsheets/d/1ib-8P9rvbIDgRNVO_Omekv3-u5b-ewHhnlkTtL8K9bk/edit?usp=sharing

mariusmotea, 2018-01-13:
x,y must have values between 0 and 256 in the final, ex for Z scale, 0 is the base of the tree and 256 is the top of the tree.

and4480, 2018-01-14:
Thank you. Your script helped greatly. Geometry ....... The brain is boiling

closed 2018-01-14 by and4480

jasoncoon, 2018-01-14:
Thanks for jumping in and helping, @mariusmotea!

For the x256, y256, and z256 arrays, they should be between 0 and **255**.  They should fit in one 8-bit byte, uint8_t, etc.

The point of the various arrays in the Map.h code are just to map the LEDs into three dimensional space in different ways.  They don't have to be exact or precise, just good enough to look good.

mariusmotea, 2018-01-14:
Just want to say that this year i had the best tree in my life with the help of this awesome project. Next year i plan to build a few 5 meters strips as Christmas gifts for my family.

and4480, 2018-01-14:
When calculating, I got a few other results, but I realized that it is possible approximately. I have 400 LEDs in the form of a spiral Archimedes, 6 turns, a cone. The formula ρ = (a / 2π) φ. But ... how to apply it? I divided the spiral into 12 semirings and calculated the coordinates. My Christmas tree was also beautiful, but some 3D models did not work correctly. I make a glass  fiber chandelier. About this:https://docviewer.yandex.ru/view/49295899/htmlimage?id=k0mn-36vinnoz9vkccuq3tdxhhliv2u6cdetpfy9lcea37ff4fav5javtxvfuatfr1wtu1oqkdv5zt1ufx33w3mlam05uvn7kqiytjd2&name=image-WjkVVxmDSQNrjaDKqV.jpg&dsid=d4419f1a84e809227ad145b5c280b0b2