evilgeniuslabs/tree-v2 #28: Sketch does not compile

vatnique66, 2019-11-24:
Hello Jason!  
When trying to compile I get this error - too many initializers for 'const uint8_t [100] {aka const unsigned char [100]}'Could you say that I have to change in the Map.h field for my number of neopixels, which is 100

jasoncoon, 2019-11-24:
Hello!  This sketch is designed for 300 LEDs, so there are 300 entries in the arrays in the Map.h file: https://github.com/evilgeniuslabs/tree-v2/blob/master/Map.h

To get it to work with 100 LEDs, you'll need to rework those arrays with just 100 entries.  Here's the Google Sheet I used to calculate the numbers for those arrays: https://docs.google.com/spreadsheets/d/1ib-8P9rvbIDgRNVO_Omekv3-u5b-ewHhnlkTtL8K9bk/edit?usp=sharing

More details, if you haven't already read through it: https://www.evilgeniuslabs.org/tree-v2#mapping

vatnique66, 2019-11-24:
Thank you Jason- I liked the effects. And I don’t even want to use it on the tree. I just wanted to use this on tape. I have two 100 leds on WS2811 and 150 on WS2812B. In Internet, I saw that you just need to specify a fixed number of neopixels in the Map.h tab. The video https://www.youtube.com/watch?v=81C8KamkQUw&list=PLI8L7aOLKSwyMUAkjRx-l0b0nm6OPWktj&index=4&t=365s  even indicates where. But even if I do everything the same way, I still get a compilation error

jasoncoon, 2019-11-24:
For more general purpose usage, try my other project that the tree is based on: https://github.com/jasoncoon/esp8266-fastled-webserver

It won't matter how many LEDs you have with it.  Tree-v2 is specifically mapped for trees with 300 LEDs.

vatnique66, 2019-11-24:
Sorry Jason. It is this project that I am trying to replicate(esp8266-fastled-webserver). It’s just that the video that I gave as an example, there was a wrong link. Because of this, I myself write in the wrong thread. Now I’ll try to see the link that you gave.

dj0vitamin, 2025-11-29:
Hi. Here the problem:

C:\Users\dj0vi\Documents\Arduino\tree-v2\tree-v2.ino:28:10: fatal error: C:\Users\dj0vi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include/ESP8266WiFi.h>: Invalid argument
   28 | #include <ESP8266WiFi.h>c:\Users\dj0vi\Documents\Arduino\libraries\FastLED-master\src\platforms\esp\new.h
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: C:\Users\dj0vi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include/ESP8266WiFi.h>: Invalid argument

ESP8266 library version latter 3.1.2

dj0vitamin, 2026-01-03:
Twinkles.h

18 return color + incrementalColor;

Give this compilation error:
Twinkles.h:18:16: error: ambiguous overload for 'operator+' (operand types are 'const CRGB' {aka 'const fl::CRGB'} and 'CRGB' {aka 'fl::CRGB'})