evilgeniuslabs/tree-v2 #35: Mapping for smaller trees.

5kull, 2020-12-06:
Could someone explain how to change the mapping for fewer amounts of diodes, I need to remove several levels from the original mapping settings.

mariusmotea, 2020-12-06:
I create the file generate_map_h.py that generate the map for strips. I think you can use it also with normal mapping without to notice something wrong in the most effects.

jasoncoon, 2020-12-06:
Yeah, count your LEDs, the number of rings/levels, and the number of LEDs per ring/level.  Edit the values at the top of the script.  Install python, then run `python generate_map_h.py`.  Copy the output, replace all of the arrays at the top of the Map.h file.

I also documented the process of mapping my tree, along with the spreadsheet I used here: https://www.evilgeniuslabs.org/tree-v2#mapping

5kull, 2020-12-29:
Thanks Guys!