evilgeniuslabs/tree-v2 #39: When's tree v3?

Picklepc, 2023-12-01:
Just pulled out my tree for the 5th year. I still love it, but the tree itself is no longer in it's prime. When's tree v3?

eziologico, 2023-12-02:
I've been using it for more than 5 years too, it's a wonderful project!
Maybe a new version with Alexa integration…

Picklepc, 2023-12-04:
I had a few things in mind:
- A timer - including dusk to dawn settings
- Music integration - react to audio
- Pattern Playlists for those who happen keep the tree up for multiple holidays.
- I added an oled to the chip and put it in an ornament. The display could have more features and animations.
- Input Button(s)
- Mqtt/GA/Alexa/Home Assistant
- A led position calculator - I used a spreadsheet to approximate, but you could probably count the length of the strands, the number of lights and how many times it wraps around to approximate the xy&z's in a simpler step.

jasoncoon, 2023-12-05:
Sorry, I don't have any plans to continue development on this, but forks & pull requests are definitely welcome. I can't wait to see what you come up with!